geekworm-com/x728 #1: Software shutdown not working

Blackcode, 2021-05-09:
When i call x728off nothing happens...
Everything is installed correctly.
 
maintenance@solarstream:~ $ x728off
X728 Shutting down...
maintenance@solarstream:~ $

geekworm-com, 2021-05-12:
Hi,
This shell script is base on Raspberrry pi OS.

Blackcode, 2021-05-16:
I am on Raspberrry pi OS.

Belroguim, 2021-05-18:
Hi,

First of all, congratulations for your work. I am very happy with several of your products.

I am having same problem than Blackcode. I have several systems running RBPi4 and x728. It doesn't matter the OS version we are using. Command "x728off" is never working. It has nothing to do with OS but x728 version. Board x728v1.3 is not able to shutdown if you install software following wiki. Meanwhile, older versions of this boards, work nice. I have tested 4 boards x728(v1.3) and 2 boards x728 (v1.2) with this consistent result.

Geekworm, please, we need help :-)

geekworm-com, 2021-05-19:
> Hi,
>
> First of all, congratulations for your work. I am very happy with several of your products.
>
> I am having same problem than Blackcode. I have several systems running RBPi4 and x728. It doesn't [...]

Hi, I have just test 2 times, no found any question.

I wonder your some question. could you give us more detail?

Maybe you can refer to here: https://wiki.geekworm.com/X728-Software

Regards,

Belroguim, 2021-05-19:
Hi,

First time I type x728 there is no feedback. Second time feddback is:
Device or resource busy

My question is... how can I fix it?

But you are going to need more details in order to figure out why is this
happenig. I am out of the office ( I am in Massachusets time ). Tomorrow I
will give you detailed information about instructions and its feedback.

Thanks and regards,
Miguel

Belroguim, 2021-05-19:
Hi,

I have partially solved it.

Attached you can find some files associated with three different tests:
Test1: Fresh OS with board v1.2. Everything works properly.
Test2: Fresh OS with board v1.3. Same PI, only x728 board was replaced.
With some workaround, it seems that "GPIO_13_Set.py" can make softshutdown.
After using this py, command x728off has started to work.
Test3: OS of our real app (OS2), several libraries and programs were added
to a vanilla OS. I guess somehow some of the modules and programs are
interfering with software needed for x728 board. You can see in the SSH3
sesion the feedback of several commands that works properly, and x728 not
working. In this case "GPIO_13_Set.py" has not the effect of solve the
problem, like in OS1.

My questions:
Do you understand what happens in test2?
Do you plan to release a new version "x728bat_v1.3.py" ? (in my case this
is not working)
I would like to fix x728 software in OS2. Do you see how I could fix it?

Thanks,
Miguel

> Hi,
>
> First time I type x728 there is no feedback. Second time feddback is:
> Device or resource busy
>
> My question is... how can I fix it?
>
> But you are going to need more details in order to figure out why is this
> happenig. I am out of the office ( I am in Massachusets time ). Tomorrow
> I will give you detailed information about instructions and its feedback.
>
> Thanks and regards,
> Miguel
>
>
>> Hi,
>>
>> First of all, congratulations for your work. I am very happy with several
>> of your products.
>>
>> I am having same problem than Blackcode. I have several systems running
>> RBPi4 and x728. It doesn't matter the OS version we are using. Command
>> "x728off" is never working. It has nothing to do with OS but x728 version.
>> Board x728v1.3 is not able to shutdown if you install software following
>> wiki. Meanwhile, older versions of this boards, work nice. I have tested 4
>> boards x728(v1.3) and 2 boards x728 (v1.2) with this consistent result.
>>
>> Geekworm, please, we need help :-)
>>
>> Hi, I have just test 2 times, no found any question.
>>
>> I wonder your some question. could you give us more detail?
>>
>> Maybe you can refer to here: https://wiki.geekworm.com/X728-Software
>>
>> Regards,

Using username "pi".
Linux raspberrypi 5.10.17-v7l+ #1403 SMP Mon Feb 22 11:33:35 GMT 2021 armv7l

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
[... 366 lines cut ...]
******************
Voltage: 3.98V
Battery:   90%
******************
^CTraceback (most recent call last):
  File "x728bat.py", line 44, in <module>
    time.sleep(2)
KeyboardInterrupt
***@***.***:~/x728 $ sudo python x728pld.py
1.Make sure your power adapter is connected
2.Disconnect and connect the power adapter to test
3.When power adapter disconnected, you will see: AC Power Loss or Power Adapter Failure
4.When power adapter disconnected, you will see: AC Power OK, Power Adapter OK
Testing Started---AC Power OK,Power Adapter OK---
---AC Power Loss OR Power Adapter Failure---
---AC Power OK,Power Adapter OK---
---AC Power Loss OR Power Adapter Failure---
^CTraceback (most recent call last):
  File "x728pld.py", line 20, in <module>
    raw_input("Testing Started")
KeyboardInterrupt
***@***.***:~/x728 $ x728off
/usr/local/bin/x728softsd.sh: line 5: echo: write error: Device or resource busy
X728 Shutting down...
***@***.***:~/x728 $

Blackcode, 2021-05-20:
Hello geekworm-com,

I have created a clean Raspberry PI installation for you on the faulty machine and exposed the SSH port. Most probably it won't work if you try to install everything and shut it down.

IP: 31.156.137.20
Port: 5555
Uname: pi
Passwd: raspberry

Let me know.

Thanks,
Dragos E.

Belroguim, 2021-05-24:
Hi,

I don't understand what you are proposing. But no worries, as I had said,
after running my script it has worked and still does.

I will update my pots next week.

One this that it could help me a is knowing if is possible to disable
restart function associated with a short push in the button. May be just
removing some part on the RB scripts. Even more impresive could be made
software shut down when user make two fast clicks on button. I don't know
how feaseble, but sure it would help more people tunning board behaviour
according theirs needs.

Thanks,
Miguel