geekworm-com/x728 #3: x728bat.py battery status issue

alpapak, 2021-07-09:
I set up and installed software. Add the batteries and full charge. Then i left my pi to discharge and then plug it to recharge it. My pi is showing full charge led status. But, if i boot my pi and run x728bat.py it is showing wrong %. In my case, since it was empty, it is showing 2% charging and voltage 4.12v.

How can i fix this?

iwanttobefreak, 2022-10-31:
Same issue.
Full charge. Power on, unplugged, shows: 
Voltage: 4.08V - Battery: 3%
The value of % grows in time and after 1h30m the value of % is 99%

katsar0v, 2026-09-18:
Hi, I'm seeing a similar discrepancy between the X728's battery LEDs and its I²C state-of-charge reading. Is there any update on the cause or a supported fix for this issue?

Setup:
- Raspberry Pi 4 running Ubuntu 26.04 ARM64.
- X728, with packaging labeled V2.5 (PCB revision not independently verified yet).
- Two matching INR18650 cells marked 3.2 Ah, newly installed.

While charging, the first battery-gauge LED is solid green and the second is flashing. The [official LED table](https://wiki.geekworm.com/X728-hardware#Fuel_gauge_-_LED_Indicator) maps that pattern to 25–50%. GPIO6 also reads LOW, indicating external power is present.

However, read-only measurements on September 18, 2026, over approximately five minutes returned:

| Sample | SOC reported by the gauge | Battery voltage |
| --- | ---: | ---: |
| Initial | 8.58% | 3.804 V |
| Later | 9.48% | 3.810 V |
| Latest | 11.29% | 3.824 V |

These percentages are read directly from the fuel gauge at I²C bus 1, address `0x36`, SOC register `0x04`, using the same byte order and `SOC / 256` calculation as the [official sample](https://github.com/geekworm-com/x728-script/blob/master/sample/x728-v2.x-bat.py). They are not calculated from voltage.

I cross-checked a raw I²C byte read against an SMBus word read with the required byte swap; both gave the same result. For example:

`SOC bytes: 0B 4B → 0x0B4B / 256 = 11.29296875%`

The initial reading was `0x0895 / 256 = 8.58203125%`. Voltage was read separately from register `0x02`. No gauge reset, calibration, or configuration writes were performed.

Is this a known fuel-gauge initialization or calibration issue after installing batteries? Are the LED levels and I²C SOC expected to disagree by this much, and which indication should be used until they agree? If there is a recommended procedure or an updated script, could you please point us to it?

Thank you!