geekworm-com/x728 #7: Not working on Raspbian OS 64 bit.

YiweiMao, 2022-02-17:
rc.local file is still modified by this software as expected but the X728 power is cut ~6 s after the button is pressed to turn it on. The blue LED fades away quickly and turns off at the same time power is cut. 

Curiously, the button can still be used to safely shutdown. 

I'm on Raspbian OS 64 bit downloaded here https://www.raspberrypi.com/software/operating-systems/
[image]

YiweiMao, 2022-02-17:
Update: May have fried the board? I was using GPIO26 as a sensor input. 

The documentation for the GPIO used by the X728 https://wiki.geekworm.com/X728-hardware#Pins_and_GPIO_used
says it does not use GPIO26 so I went ahead and used it assuming it was accurate. 
[image]
It includes a NEW for V2.1 but does not indicate the other changes in V2.0 where
PIN33 (GPIO13) is replaced by the PIN37 (GPIO26) to avoid the I2S interface from v1.3 to v2.0.

Please fix the documentation so no one else makes the mistake.

rkolbi, 2023-01-19:
> sudo bash x728-v2.1.sh

Do an 'ls' to make sure you are in the correct directory and can see the file first, if so - try:
sudo ./x728-v2.1.sh

keeper18, 2023-01-20:
> > sudo bash x728-v2.1.sh
> 
> Do an 'ls' to make sure you are in the correct directory and can see the file first, if so - try: sudo ./x728-v2.1.sh

Thanks for help.I'm in home directory due to ls command:)now I try another manual from http://www.suptronics.com/miniPCkits/x728_v2.2-software.html similar to first one and I can totally shutdown RPI by  "x728off " command:)My ups board is V2.0.

rkolbi, 2023-01-27:
It looks like section 3.1 on that page tells you how to add it to cron. Since the python scripts outputs text, you cannot simply add it to the end of rc.local.

keeper18, 2023-01-27:
> It looks like section 3.1 on that page tells you how to add it to cron. Since the python scripts outputs text, you cannot simply add it to the end of rc.local.

Yes I know I downloaded main/bat.py but they don't load when raspbian starts...

rkolbi, 2023-01-28:
Did you execute the bash script that installs the power button files? Maybe this will help - I have made a simple UPS python script that is ran at boot, can see it here: https://raw.githubusercontent.com/rkolbi/voron2.4/main/Tap-Tap-Sensorless-Huvud/rPi%20Files/X728_UPS.txt

YiweiMao, 2023-01-29:
I used systemd to enable /etc/rc-local and get the scripts running on startup. Here's my documentation on my install process
https://openhsi.github.io/openhsi/tutorial_installing_linux.html#4.5-Installing-the-Battery-Hat-software

keeper18, 2023-01-29:
> Did you execute the bash script that installs the power button files? Maybe this will help - I have made a simple UPS python script that is ran at boot, can see it here: https://raw.githubusercontent.com/rkolbi/voron2.4/main/Tap-Tap-Sensorless-Huvud/rPi%20Files/X728_UPS.txt

Yes.All script works except buzzer because v2.0 have no buzzer.I can reboot hold 2s, shut down press 4s and force shutdown pressing long...but till reboot raspbian.How to load script showing voltage/charge automatically because there is no bar icon and battery level leds are in the box?