geekworm-com/x728 #9: Syslog Filled With Status Messages

W4JEW, 2022-05-07:
The x728pwr.sh script is causing excessive log messages to be written to /var/log/syslog

It appears to check the power input level multiple times per second and writing a message to syslog each time.

This is terrible behavior in that it causes a tremendous amount of disk usage and also makes it next to impossible to view other syslog messages unrelated to the Geekworm product.

Please provide a method to disable logging and/or decrease the rate at which the script is polling for status.

This is what I see in the syslog messages:

```
May  7 13:30:24 harmonic x728pwr.sh[423]: Power Input Okay
May  7 13:30:24 harmonic x728pwr.sh[423]: Battery Voltage: 4.18V
May  7 13:30:24 harmonic x728pwr.sh[423]: Battery Capacity: 100.03%
May  7 13:30:24 harmonic x728pwr.sh[423]: Power Input Okay
May  7 13:30:24 harmonic x728pwr.sh[423]: Battery Voltage: 4.18V
May  7 13:30:24 harmonic x728pwr.sh[423]: Battery Capacity: 100.03%
May  7 13:30:24 harmonic x728pwr.sh[423]: Power Input Okay
May  7 13:30:24 harmonic x728pwr.sh[423]: Battery Voltage: 4.18V
May  7 13:30:24 harmonic x728pwr.sh[423]: Battery Capacity: 100.03%
May  7 13:30:24 harmonic x728pwr.sh[423]: Power Input Okay
May  7 13:30:25 harmonic x728pwr.sh[423]: Battery Voltage: 4.18V
May  7 13:30:25 harmonic x728pwr.sh[423]: Battery Capacity: 100.03%
May  7 13:30:25 harmonic x728pwr.sh[423]: Power Input Okay
May  7 13:30:25 harmonic x728pwr.sh[423]: Battery Voltage: 4.18V
May  7 13:30:25 harmonic x728pwr.sh[423]: Battery Capacity: 100.03%
May  7 13:30:25 harmonic x728pwr.sh[423]: Power Input Okay
May  7 13:30:25 harmonic x728pwr.sh[423]: Battery Voltage: 4.18V
May  7 13:30:25 harmonic x728pwr.sh[423]: Battery Capacity: 100.03%
May  7 13:30:25 harmonic x728pwr.sh[423]: Power Input Okay
May  7 13:30:25 harmonic x728pwr.sh[423]: Battery Voltage: 4.18V
May  7 13:30:25 harmonic x728pwr.sh[423]: Battery Capacity: 100.03%
May  7 13:30:25 harmonic x728pwr.sh[423]: Power Input Okay
May  7 13:30:25 harmonic x728pwr.sh[423]: Battery Voltage: 4.18V
May  7 13:30:25 harmonic x728pwr.sh[423]: Battery Capacity: 100.03%
May  7 13:30:25 harmonic x728pwr.sh[423]: Power Input Okay
May  7 13:30:25 harmonic x728pwr.sh[423]: Battery Voltage: 4.18V
May  7 13:30:25 harmonic x728pwr.sh[423]: Battery Capacity: 100.03%
May  7 13:30:25 harmonic x728pwr.sh[423]: Power Input Okay
May  7 13:30:25 harmonic x728pwr.sh[423]: Battery Voltage: 4.18V
May  7 13:30:25 harmonic x728pwr.sh[423]: Battery Capacity: 100.03%
May  7 13:30:25 harmonic x728pwr.sh[423]: Power Input Okay
May  7 13:30:25 harmonic x728pwr.sh[423]: Battery Voltage: 4.18V
May  7 13:30:25 harmonic x728pwr.sh[423]: Battery Capacity: 100.03%
May  7 13:30:25 harmonic x728pwr.sh[423]: Power Input Okay
May  7 13:30:25 harmonic x728pwr.sh[423]: Battery Voltage: 4.18V
May  7 13:30:25 harmonic x728pwr.sh[423]: Battery Capacity: 100.03%
May  7 13:30:25 harmonic x728pwr.sh[423]: Power Input Okay
May  7 13:30:25 harmonic x728pwr.sh[423]: Battery Voltage: 4.18V
May  7 13:30:25 harmonic x728pwr.sh[423]: Battery Capacity: 100.03%
May  7 13:30:25 harmonic x728pwr.sh[423]: Power Input Okay
May  7 13:30:26 harmonic x728pwr.sh[423]: Battery Voltage: 4.18V
May  7 13:30:26 harmonic x728pwr.sh[423]: Battery Capacity: 100.03%
May  7 13:30:26 harmonic x728pwr.sh[423]: Power Input Okay
May  7 13:30:26 harmonic x728pwr.sh[423]: Battery Voltage: 4.18V
May  7 13:30:26 harmonic x728pwr.sh[423]: Battery Capacity: 100.03%
May  7 13:30:26 harmonic x728pwr.sh[423]: Power Input Okay
May  7 13:30:26 harmonic x728pwr.sh[423]: Battery Voltage: 4.18V
May  7 13:30:26 harmonic x728pwr.sh[423]: Battery Capacity: 100.03%
May  7 13:30:26 harmonic x728pwr.sh[423]: Power Input Okay
May  7 13:30:26 harmonic x728pwr.sh[423]: Battery Voltage: 4.18V
May  7 13:30:26 harmonic x728pwr.sh[423]: Battery Capacity: 100.03%
May  7 13:30:26 harmonic x728pwr.sh[423]: Power Input Okay
May  7 13:30:26 harmonic x728pwr.sh[423]: Battery Voltage: 4.18V
May  7 13:30:26 harmonic x728pwr.sh[423]: Battery Capacity: 100.03%
May  7 13:30:26 harmonic x728pwr.sh[423]: Power Input Okay
May  7 13:30:26 harmonic x728pwr.sh[423]: Battery Voltage: 4.18V
May  7 13:30:26 harmonic x728pwr.sh[423]: Battery Capacity: 100.03%
May  7 13:30:26 harmonic x728pwr.sh[423]: Power Input Okay
May  7 13:30:26 harmonic x728pwr.sh[423]: Battery Voltage: 4.18V
[... 140 lines cut ...]
May  7 13:30:31 harmonic x728pwr.sh[423]: Battery Voltage: 4.18V
May  7 13:30:31 harmonic x728pwr.sh[423]: Battery Capacity: 100.03%
May  7 13:30:31 harmonic x728pwr.sh[423]: Power Input Okay
May  7 13:30:31 harmonic x728pwr.sh[423]: Battery Voltage: 4.18V
May  7 13:30:31 harmonic x728pwr.sh[423]: Battery Capacity: 100.03%
May  7 13:30:31 harmonic x728pwr.sh[423]: Power Input Okay
May  7 13:30:31 harmonic x728pwr.sh[423]: Battery Voltage: 4.18V
May  7 13:30:31 harmonic x728pwr.sh[423]: Battery Capacity: 100.03%
May  7 13:30:31 harmonic x728pwr.sh[423]: Power Input Okay
May  7 13:30:32 harmonic x728pwr.sh[423]: Battery Voltage: 4.18V
May  7 13:30:32 harmonic x728pwr.sh[423]: Battery Capacity: 100.03%
May  7 13:30:32 harmonic x728pwr.sh[423]: Power Input Okay
May  7 13:30:32 harmonic x728pwr.sh[423]: Battery Voltage: 4.18V
May  7 13:30:32 harmonic x728pwr.sh[423]: Battery Capacity: 100.03%
May  7 13:30:32 harmonic x728pwr.sh[423]: Power Input Okay
May  7 13:30:32 harmonic x728pwr.sh[423]: Battery Voltage: 4.18V
May  7 13:30:32 harmonic x728pwr.sh[423]: Battery Capacity: 100.03%
May  7 13:30:32 harmonic x728pwr.sh[423]: Power Input Okay
May  7 13:30:32 harmonic x728pwr.sh[423]: Battery Voltage: 4.18V
May  7 13:30:32 harmonic x728pwr.sh[423]: Battery Capacity: 100.03%
May  7 13:30:32 harmonic x728pwr.sh[423]: Power Input Okay
May  7 13:30:32 harmonic x728pwr.sh[423]: Battery Voltage: 4.18V
May  7 13:30:32 harmonic x728pwr.sh[423]: Battery Capacity: 100.03%
May  7 13:30:32 harmonic x728pwr.sh[423]: Power Input Okay
```

geekworm-com, 2022-05-09:
Hi, 

The log file shows that you are constantly reading the battery, you can't run the script to read the battery in the background. Wondering if you did secondary development of the script?

W4JEW, 2022-05-11:
Your documentation is extremely confusing. I did not do any additional development to the script myself.  Thankfully there are other customers that have taken the scripts you provided and turned them into something that is easier to work with.

I used the following repository:

https://github.com/federicobriata/x728-script

It looks like Federico creates a service with the name "x728pwr.service"

I also have another service file: x728-shutdown.service

I believe this is the service that you provide that calls x728v2-asd.py

```
[Unit]
Description=Geekworm x728 Shutdown Service
After=network.target

[Service]
Type=simple
StandardOutput=null
WorkingDirectory=/usr/local/bin
User=root
Restart=always
RestartSec=3
ExecStart=/usr/bin/python3 /usr/local/bin/x728v2-asd.py
ExecReload=/bin/kill -2 $MAINPID
KillMode=process

[Install]
WantedBy=multi-user.target
```

Then x728v2-asd.py contains:

```
#!/usr/bin/env python
# We want to automatically shut down when the battery voltage is lower than a certain specified value, is it possible?
# Please run the 'python3 /home/pi/x728/x728v2-asd.py &' to get this function

import struct
import smbus
import sys
import time
import RPi.GPIO as GPIO

GPIO_PORT=26
I2C_ADDR=0x36

GPIO.setmode(GPIO.BCM)
GPIO.setup(GPIO_PORT, GPIO.OUT)
GPIO.setwarnings(False)

def readVoltage(bus):

     address = I2C_ADDR
     read = bus.read_word_data(address, 2)
     swapped = struct.unpack("<H", struct.pack(">H", read))[0]
     voltage = swapped * 1.25 /1000/16
     return voltage

def readCapacity(bus):

     address = I2C_ADDR
     read = bus.read_word_data(address, 4)
     swapped = struct.unpack("<H", struct.pack(">H", read))[0]
     capacity = swapped/256
     return capacity

bus = smbus.SMBus(1)

while True:

 print ("******************")
 print ("Voltage:%5.2fV" % readVoltage(bus))
 print ("Battery:%5i%%" % readCapacity(bus))

 if readCapacity(bus) == 100:
         print ("Battery FULL")

 if readCapacity(bus) < 20:
         print ("Battery Low")

#Set battery low voltage to shut down
 if readVoltage(bus) < 3.00:
         print ("Battery LOW!!!")
         print ("Shutdown in 10 seconds")
         time.sleep(10)
         GPIO.output(GPIO_PORT, GPIO.HIGH)
         time.sleep(3)
         GPIO.output(GPIO_PORT, GPIO.LOW)

 time.sleep(2)
```

I had both services enabled. I removed Federico's x728pwr.service service and left the x728-shutdown.service enabled. That seems to be working better, but I have not been able to perform a proper test as the device is running in a remote location that I cannot easily access if things break.

closed 2022-05-11 by W4JEW