geekworm-com/x728 #14: lack of overcharge protection?

markuman, 2025-04-05:
After 48h on power, the voltage is now at 4.24V and the capacity is returned at 106%.  
It looks like there is no or no sufficient overload protection.

markuman, 2025-04-05:
https://wiki.geekworm.com/X728-script#How_to_Use_GPIO16_to_control_battery_charging

```
GPIO=16

# Read the /sys/kernel/debug/gpio file to get the actual GPIO pin
GPIO=$(cat /sys/kernel/debug/gpio | grep "GPIO$GPIO" | awk -F'gpio-' '{print $2}' | awk -F' ' '{print $1}')
echo "$GPIO" > /sys/class/gpio$GPIO/export
echo "out" > /sys/class/gpio/gpio$GPIO/direction
```

this results in 

```
echo "$GPIO" > /sys/class/gpio$GPIO/export
bash: /sys/class/gpio528/export: Datei oder Verzeichnis nicht gefunden # not found
```

geekworm-com, 2025-04-07:
Hi，

Refer to [X728-script](https://github.com/geekworm-com/x728-script)

This one is abandoned

markuman, 2025-04-07:
> Hi，
> 
> Refer to [X728-script](https://github.com/geekworm-com/x728-script)

there is no charging example: https://github.com/search?q=repo%3Ageekworm-com%2Fx728-script%20charging&type=code